inseven/statuspanel #429: web sim fixups

pvinis, 2023-03-12:
- use uuid
- cycle through images

jbmorley, 2023-03-12:
Let's make the 'highlight' color #7FFFD4.

jbmorley, 2023-03-12:
Let's unify the background color of the website and simulator too. Maybe we support dark mode in both?

jbmorley, 2023-03-12:
How about automatically checking for an update when we start so the user doesn't have to initiate it?

jbmorley, 2023-03-12:
Don't show the QR code when we know we've got an update.

jbmorley, 2023-03-12:
Pretty sure we're not going to fix this bug all at once, so, while I'm thinking about it--I'd love to figure out a way to cleanly merge the navigation links from the top of the website with this. Or, at the very least, implement them twice, and style them similarly so you don't randomly get stuck in the simulator.

jbmorley, 2023-03-12:
Maybe we can re-fetch when the window gets focus again?

closed 2026-10-06 by jbmorley